syssi/esphome-jnge-mppt-controller #28: Donation Option

MasterCATZ, 2022-04-30:
Thank you for creating this, I looked around but could not find any paypal / buy coffee / beer options  

I had bought over 20x of these units with wifi , but could NEVER get their damn software to work 
as I am monitoring every single battery cell this works out good for me as I can just use my esp32 to collect the data and send it right to "Home Assistant"

syssi, 2022-05-01:
> Thank you for creating this, I looked around but could not find any paypal / buy coffee / beer options

You can find a buymeacoffee.com button at the top of the README.md: https://www.buymeacoffee.com/syssi

To use the `api` component instead of `mqtt` you have to replace this section:

```
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client
```

with

```
api:
```

See https://esphome.io/components/api.html

Feel free to ask further questions if you have trouble to use this component.

closed 2022-05-22 by syssi

MasterCATZ, 2022-05-27:
> 
> You can find a buymeacoffee.com button at the top of the README.md: https://www.buymeacoffee.com/syssi
> 

I know I'm blind but I can still not find any links on the README.md page

syssi, 2022-05-27:
Thanks for your donation! I will enjoy the ☕!