syssi/esphome-jnge-mppt-controller #30: JN-MPPT-AL Buck Solar Charge Controller

MasterCATZ, 2022-05-27:
I have 30A chargers , I did buy them with wifi modules but could never get them to work 

I also can not seem to get this to work either 

anyway to test the MPPT RJ45 ports that they are actually working ? or could they have changed the pinout ? 

on the Elfin-EW11A
Pin 8 GND Orange X Pin 3 Orange
Pin 7 VCC Orange Stripe X Empty 
Pin 6  B- Green X Pin 4 Green
Pin 5 A+ Green Stripe x Empty (?)
Pin 4  Reload Brown X Pin 1 Brown
Pin 3 Brown Stripe X Empty
Pin 2  Blue X Pin 2 Blue (?)
Pin 1  Blue Stripe X  Empty

now looking at  this it seems they might have sent me wrong cables no VCC / A+ connection

syssi, 2022-05-27:
Could you do two measurements:

1. Please measure the voltage between Pin 6 and Pin 8. You have to make sure the device is connected to a battery.
2. If there is no voltage at step 1 please measure the voltage between Pin 7 and Pin 8.

I'm aware of two pinouts of the RJ485 jack. Please compare the pinout of the JN-MPPT README with the JN-WIND README.

syssi, 2022-05-27:
It looks like your WiFi module has the JN-WIND pinouts (may be the "new pinout"). Your solar charger has the JN-MPPT (may be the "old pinout"). If you prepare a new cable you should be able to get both solutions up and running.

MasterCATZ, 2022-05-27:
using 568B

Green and Brown 
Green -13.8v Brown +13.8v

Brown stripe and Brown

syssi, 2022-05-27:
This is what I've understood: 

You have measured the voltage between pin 6 and 8 and it's 13.8V. Could you measure the voltage between pin 5 (GND) and 6 (GND). It should be 0V.

If the above is true we have identified this pinout:

https://github.com/syssi/esphome-jnge-mppt-controller/blob/main/jnge_mppt_controller.md#rj45-jack

```
Pin 1: B-
Pin 2: B-
Pin 3: B-
Pin 4: B-
Pin 5: GND
Pin 6: GND
Pin 7: VCC
Pin 8: VCC
```

The next step is to attach the RS485 converter to the RX/TX pins of your ESP:

https://raw.githubusercontent.com/syssi/esphome-jnge-mppt-controller/main/images/circuit.jpg

The TX pin (of the RS485 converter) must blink periodically. If the RX pin blinks periodically you have to swap RX/TX. 

The RS485 lines must not be crossed. You have to attach "B-" (orange-white) to "B-" of the converter and "A+" (green-white) to "A+".

If everything is ready please provide a photo of your circuit/wiring.

MasterCATZ, 2022-05-27:
yes the RS485 does blink

syssi, 2022-05-27:
It's important the TX LED of the RS485 does blink. ;-) As next step I would like to see your wiring.

MasterCATZ, 2022-05-27:
https://drive.google.com/drive/folders/17dD09No1H9q71I7gLnj-KpSR9HYUSzxJ?usp=sharing

syssi, 2022-05-27:
I will try to replicate your setup to provide some details. The videos doesn't look good. I assume your setup crashs/reboots periodically.

MasterCATZ, 2022-05-27:
no just wifi losses 

its just for testing , I have been waiting for shelves to arrive since last year ... 
Currently it is 
RS485 to TTL RX to RX on ESP8266
RS485 to TTL TX to TX on ESP8266

are you positive is should not be 

RS485 to TTL RX to TX on ESP8266
RS485 to TTL TX to RX on ESP8266

syssi, 2022-05-27:
I've prepared a small demo. I've used this config example:

```
substitutions:
  name: jn-mppt
  status: "${name} status"
  config: "${name} config"
  external_components_source: github://syssi/esphome-jnge-mppt-controller@main
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

api:
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

uart:
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jnge_mppt_controller:
  enable_fake_traffic: false
  suppress_battery_temperature_errors: false
  update_interval: 2s

binary_sensor:
  - platform: jnge_mppt_controller
    fan_running:
      name: "${status} fan running"
    load_detected:
      name: "${status} load detected"
    load:
      name: "${status} load"
    charging:
      name: "${status} charging"

sensor:
  - platform: jnge_mppt_controller
    # status
    firmware_version:
      name: "${status} firmware version"
    pv_voltage:
      name: "${status} pv voltage"
    battery_voltage:
      name: "${status} battery voltage"
    channel_a_current:
      name: "${status} channel a current"
    channel_b_current:
      name: "${status} channel b current"
    charging_current:
      name: "${status} charging current"
[... 113 lines cut ...]
    battery_undervoltage:
      name: "${config} battery undervoltage"
    equalization_charging_time:
      name: "${config} equalization charging time"
    improve_charging_time:
      name: "${config} improve charging time"
    light_control_on_voltage:
      name: "${config} light control on voltage"
    light_control_off_voltage:
      name: "${config} light control off voltage"
    light_control_on_period_1:
      name: "${config} light control on period 1"
    light_control_on_period_2:
      name: "${config} light control on period 2"
    load_turn_off_time:
      name: "${config} load turn off time"
    battery_strings:
      name: "${config} battery strings"
```

I've used GPIO4 and GPIO5 (software serial) on purpose to be able to receive log message via UART0 (GPIO1, GPIO3). 

Please have a look at this video. The `update_interval` of the component is `2 seconds` therefore the component transmits one request per 2 seconds. The TX LED must blink every 2 seconds. Please compare this behavior with your TX LED. Something wrong is going on here with your setup. 

https://user-images.githubusercontent.com/2735933/170671014-d9f35611-1045-4d30-86ce-66e932a52b12.mp4

syssi, 2022-05-27:
As next step I've attached a JN-MPPT-Mini to the setup powered by a small 12V battery:

[image]
[image]
[image]

syssi, 2022-05-27:
If the JN-MPPT responds you should see something like this:

https://user-images.githubusercontent.com/2735933/170672525-2c48eccd-8171-412c-ae4c-5001b54f8923.mp4

Plus some debug output at the log:

```
[11:04:43][D][sensor:124]: 'jn-mppt status firmware version': Sending state 4.00000  with 0 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status pv voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status battery voltage': Sending state 12.20000 V with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status channel a current': Sending state 0.00000 A with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status channel b current': Sending state 0.00000 A with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status charging current': Sending state 0.00000 A with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status charger temperature': Sending state 22.00000 °C with 0 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status charging power': Sending state 0.00000 W with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status discharge power': Sending state 0.00000 W with 1 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status battery capacity': Sending state 40.00000 % with 0 decimals of accuracy
[11:04:43][D][sensor:124]: 'jn-mppt status operation mode id': Sending state 0.00000  with 0 decimals of accuracy
[11:04:43][D][text_sensor:067]: 'jn-mppt status operation mode': Sending state 'Not charging'
[11:04:43][D][binary_sensor:036]: 'jn-mppt status load detected': Sending state ON
[11:04:43][D][sensor:124]: 'jn-mppt status discharge current': Sending state 0.00000 A with 1 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status discharge times': Sending state 0.00000  with 0 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status total charging energy': Sending state 0.40000 kWh with 1 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status total discharge energy': Sending state 0.00000 kWh with 1 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status battery temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status error bitmask': Sending state 24576.00000  with 0 decimals of accuracy
[11:04:44][D][text_sensor:067]: 'jn-mppt status errors': Sending state 'Battery temperature sensor failure;PV array undervoltage'
[11:04:44][D][sensor:124]: 'jn-mppt status controller voltage level': Sending state 12.00000 V with 0 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status pv standby voltage': Sending state 11.30000 V with 1 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status controller current level': Sending state 30.00000 A with 0 decimals of accuracy
[11:04:44][D][sensor:124]: 'jn-mppt status battery strings': Sending state 1.00000  with 0 decimals of accuracy
[11:04:44][D][text_sensor:067]: 'jn-mppt status battery type': Sending state 'Lead-Acid'
[11:04:44][D][binary_sensor:036]: 'jn-mppt status charging': Sending state ON
[11:04:44][D][binary_sensor:036]: 'jn-mppt status fan running': Sending state OFF
[11:04:44][D][sensor:124]: 'jn-mppt status battery temperature compensation voltage point': Sending state 0.00000 mV with 0 decimals of accuracy
[11:04:44][D][binary_sensor:036]: 'jn-mppt status load': Sending state ON
```

MasterCATZ, 2022-05-27:
[d1minibackup.txt](https://github.com/syssi/esphome-jnge-mppt-controller/files/8785570/d1minibackup.txt)

Backup file for me and going to try yours 
I tried swapping the tx / rx again all it did was change which led went so both leds do work , its just both do not run ... 

and bugga forgotten to slip in the networking details

MasterCATZ, 2022-05-27:
Ok D1 and D2 are a win 

now to find what my issue was 

THANK YOU 

I'll donate soon

syssi, 2022-05-27:
You are welcome! No donation required.

MasterCATZ, 2022-05-27:
well the esphome  terminal is showing data but home assistant is not and it did before using the fake stuff ... I'll keep digging around at least I know its not hardware now

syssi, 2022-05-27:
The "esphome node name" has changed because of my yaml. You have to remove the (old node) and add the new one. Or you have to change the node name back to the registered one.

MasterCATZ, 2022-05-27:
yep gone back to my config all working now with just RX/TX pins changed  

```
  tx_pin: GPIO4 #D2
  rx_pin: GPIO5 #D1
```

closed 2022-05-27 by MasterCATZ

syssi, 2022-05-27:
Last question out of curiosity: Where are you from?

MasterCATZ, 2022-05-29:
Australia QLD