syssi/esphome-jnge-mppt-controller #31: jnge_select.h: No such file or directory

MasterCATZ, 2022-05-27:
possibly one of those changes in the last 9 hours is stopping the build ? 

```
INFO Reading configuration /config/esphome/esphome-web-410164.yaml...
INFO Updating https://github.com/syssi/esphome-jnge-mppt-controller.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-410164 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/esphome-web-410164/.pioenvs/esphome-web-410164/src/esphome/components/jnge_mppt_controller/jnge_mppt_controller.cpp.o
Compiling /data/esphome-web-410164/.pioenvs/esphome-web-410164/src/esphome/components/jnge_mppt_controller/number/jnge_number.cpp.o
In file included from src/esphome/components/jnge_mppt_controller/jnge_mppt_controller.cpp:1:
src/esphome/components/jnge_mppt_controller/jnge_mppt_controller.h:8:10: fatal error: esphome/components/jnge_mppt_controller/select/jnge_select.h: No such file or directory
    8 | #include "esphome/components/jnge_mppt_controller/select/jnge_select.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/esphome-web-410164/.pioenvs/esphome-web-410164/src/esphome/components/jnge_mppt_controller/jnge_mppt_controller.cpp.o] Error 1
In file included from src/esphome/components/jnge_mppt_controller/number/jnge_number.h:3,
                 from src/esphome/components/jnge_mppt_controller/number/jnge_number.cpp:1:
src/esphome/components/jnge_mppt_controller/number/../jnge_mppt_controller.h:8:10: fatal error: esphome/components/jnge_mppt_controller/select/jnge_select.h: No such file or directory
    8 | #include "esphome/components/jnge_mppt_controller/select/jnge_select.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/esphome-web-410164/.pioenvs/esphome-web-410164/src/esphome/components/jnge_mppt_controller/number/jnge_number.cpp.o] Error 1
========================== [FAILED] Took 1.20 seconds ==========================

```

syssi, 2022-05-28:
Sometimes a clean build is required:

```
esphome clean yourconfig.yaml
esphome run yourconfig.yaml
```

Does this solve the build issue?

syssi, 2022-05-28:
Another idea: It's possible the build is broken now if this (new) entity isn't added to the config. If the clean build doesn't solve the issue please try to add this entity to your config and (clean) build again:

```
select:
  - platform: jnge_mppt_controller
    battery_type:
      name: "${config} battery type"
      optionsmap:
        "Lead Acid": 0
        "Gel": 1
        "Ternary Lithium": 2
        "LiFePo4": 3
        "Custom": 4
```

syssi, 2022-05-28:
I was able to reproduce the issue and my last guess was right. By my change yesterday this lines are required now

```
select:
  - platform: jnge_mppt_controller
```

Sorry for the mess. I will try to make the entity/platform/line optional again.

syssi, 2022-05-28:
I have an idea what needs to be done. I will refactor the feature later the day.

closed 2022-05-28 by syssi

MasterCATZ, 2022-05-28:
Confirmed Fixed , Thanks 

I'll just keep pulling after I verify they work so I can just switch over to my fork if something does break again :D

PR #32 (2022-05-28): Refactor select entity
Fixes: #31
commit 28e9e320 (2022-05-28): Refactor select entity (#32)

Fixes: #31