syssi/esphome-jnge-mppt-controller #34: Solafans MPPT

woodie2, 2022-10-24:
Hello, is it possible to support the following MPPT? Do the communication protocol look similar?

[Translated copy of MPPT communication protocol.pdf](https://github.com/syssi/esphome-jnge-mppt-controller/files/9849922/Translated.copy.of.MPPT.communication.protocol.pdf)

https://www.alibaba.com/product-detail/Advanced-Technology-96V-45A-MPPT-Solar_1600462053979.html

syssi, 2022-10-24:
The protocol is different but the skeleton of this component could be used to implement this inverter / protocol.

syssi, 2023-02-05:
Did you find a solution to talk to the device? Do you own the device already?

closed 2023-02-11 by syssi

woodie2, 2023-02-12:
> Did you find a solution to talk to the device? Do you own the device already?

yes I have the device, and it is working with their software, but I don't have the time to implement the new protocol here.

JasonL5, 2024-11-28:
Have you ever done the protocol yet