syssi/esphome-jnge-mppt-controller #35: New rj45

woodie2, 2022-12-07:
jnge has changed the pins at their mppts:

[JN-MPPT-L user manual-（10A_120A）Buck Series Solar Charge Controller.pdf](https://github.com/syssi/esphome-jnge-mppt-controller/files/10176303/JN-MPPT-L.user.manual-.10A_120A.Buck.Series.Solar.Charge.Controller.pdf)

woodie2, 2022-12-07:
it works after connecting the correct pins, but I have a lot of CRC Check fails, is it normal?

[16:03:22][D][uart_debug:114]: <<< 00:01:00:00:00:00:00:00:00:05:00:00:00:00:7F:D0:20:00:00:02:03:76:00:07:00:10:00:03:00:00:00:00:00:00:00:01:06:41:EF:FF:EE:EF:FF:FF:FF:FD:FF:06:04:3A:03:EC:03:7D:02:0B:00:00:00:00:00:00:00:0A:00:00:00:00:00:28:00:01:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:20:00:00:02:03:75:00:07:00:10:00:03:00:01:00:00:00:00:00:01:45:40
[16:03:22][D][uart_debug:114]: >>> 06:03:10:24:00:19:C1:7C
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! 54BC!=7D03
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! 30D7!=00
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=A00
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! 1408!=01
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[16:03:22][W][jnge_modbus:089]: JngeModbus CRC Check failed! 9072!=00
[16:03:23][W][jnge_modbus:089]: JngeModbus CRC Check failed! F969!=200
[16:03:23][W][jnge_modbus:089]: JngeModbus CRC Check failed! 50A7!=07
[16:03:23][W][jnge_modbus:089]: JngeModbus CRC Check failed! CF01!=00
[16:03:23][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=4045
[16:03:23][D][uart_debug:114]: <<< FF:FF:FF:FF:EE:FF:FF:FF:FF:06:04:3A:03:EC:03:7D:02:0B:00:00:00:00:00:00:00:0A:00:00:00:00:00:28:00:01:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:20:00:00:02:03:75:00:07:00:00:00:03:00:01:00:00:00:00:00:00:45:40:DF:FF:FF:FF:FF:FF:FF:FF:FF:DB:FF:FF:FF:FF:FF:7F:FF:FF:FF:06:03:32:02:70:02:50:02:50:02:80:02:40:02:10:02:20:03:C0:06:00:01:C0:00:02:00:01:00:03:00:06:00:32:00:3C:00:18:00:00
[16:03:24][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[16:03:26][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 5070!=00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 4800!=10
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 9FC1!=F727
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 54BC!=7D03
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 30D7!=00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=A00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 1408!=01
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 9072!=00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 3970!=200
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 50A7!=07
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! CF02!=00
[16:03:26][W][jnge_modbus:089]: JngeModbus CRC Check failed! 47C0!=FF40
[16:03:26][D][uart_debug:114]: <<< 00:01:00:00:00:01:00:01:00:10:00:03:00:01:95:27:F7:FF:FF:FF:FF:FF:FF:FD:FE:FB:EF:FF:06:04:3A:03:EC:03:7D:02:0B:00:00:00:00:00:00:00:0A:00:00:00:00:00:28:00:01:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:00:00:00:02:03:75:00:07:00:10:00:03:00:00:00:00:00:00:00:01:45:40:FF:FF:FF:FE:FF:FF:FF:FF:BF:06:04:3A:03:EC:03:7C:02:09:00:00:00:00:00:00:00:0A:00:00:00

syssi, 2022-12-07:
If we ignore the CRC errors is the component able to decode the frames in general? How long is your cable? Could you provide some more traffic? I would like to compare the frames.

woodie2, 2022-12-07:
the cable is about 1m. yes the  component able to decode the frames. I see pv and battery voltage and so on. perhabs it's normal to see CRC errors, I only want to tell you. It could also be a new frame with new information. firmaware version is shown: 1004

syssi, 2022-12-07:
Could you provide a bigger capture? May be we are able to identify a pattern.

woodie2, 2022-12-07:
0:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:43:F0
[19:42:29][D][uart_debug:114]: >>> 06:03:10:24:00:19:C1:7C
[19:42:29][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:6F:FF:7F:FD:FF:FF:FF:FF:FF:FF:FB:FF:BF:FF:BF:7F:FF:FF
[19:42:29][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[19:42:30][W][jnge_modbus:089]: JngeModbus CRC Check failed! 1B3D!=5002
[19:42:30][W][jnge_modbus:089]: JngeModbus CRC Check failed! 15D!=240
[19:42:30][W][jnge_modbus:089]: JngeModbus CRC Check failed! 8498!=D0
[19:42:30][D][uart_debug:114]: <<< FF:FF:FF:FE:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:7F:FF:FB:FF:FE:FE:FF:FF:FF:06:03:32:02:70:02:50:02:50:02:40:02:40:02:10:02:20:01:C0:02:00:01:C0:00:02:00:01:00:03:00:06:00:32:00:3C:00:18:00:00:00:01:30:F7:DF:FF:FF:D5:95:17:D0:00:01:95:27:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:F7:FF:FE:FF:FF:FF
[19:42:30][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[19:42:32][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[19:42:33][W][jnge_modbus:089]: JngeModbus CRC Check failed! A9BC!=A8BC
[19:42:33][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FB:FF:FF:DF:BF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:BF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:A6:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:BC:A8:EF:FF:FF:FE:FF:FF:FF:FF:FF
[19:42:34][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! 54BC!=AA00
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! A0D6!=00
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=900
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! C8F1!=100
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! 1E00!=60D0
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! A1C4!=600
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! 7C!=03
[19:42:35][W][jnge_modbus:089]: JngeModbus CRC Check failed! 20!=00
[19:42:35][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:AA:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:06:00:10:00:03:00:01:00:00:00:00:00:01:42:1A:FF:F7:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FB:FE
[19:42:36][D][uart_debug:114]: >>> 06:04:10:00:00:1D:35:74
[19:42:37][D][sensor:127]: 'jn-mppt status firmware version': Sending state 1004.00000  with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status pv voltage': Sending state 16.10000 V with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status battery voltage': Sending state 52.20000 V with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status channel a current': Sending state 0.00000 A with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status channel b current': Sending state 0.00000 A with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status charging current': Sending state 0.00000 A with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status charger temperature': Sending state 9.00000 °C with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status charging power': Sending state 0.00000 W with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status discharge power': Sending state 0.00000 W with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status battery capacity': Sending state 40.00000 % with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status operation mode id': Sending state 0.00000  with 0 decimals of accuracy
[19:42:37][D][text_sensor:067]: 'jn-mppt status operation mode': Sending state 'Not charging'
[19:42:37][D][sensor:127]: 'jn-mppt status discharge current': Sending state 0.00000 A with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status discharge times': Sending state 0.00000  with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status total charging energy': Sending state 0.50000 kWh with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status total discharge energy': Sending state 0.00000 kWh with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status battery temperature': Sending state 65488.00000 °C with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status error bitmask': Sending state 24576.00000  with 0 decimals of accuracy
[19:42:37][D][text_sensor:067]: 'jn-mppt status errors': Sending state 'Battery temperature sensor failure;PV array undervoltage'
[19:42:37][D][sensor:127]: 'jn-mppt status controller voltage level': Sending state 48.00000 V with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status pv standby voltage': Sending state 51.50000 V with 1 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status controller current level': Sending state 50.00000 A with 0 decimals of accuracy
[19:42:37][D][sensor:127]: 'jn-mppt status battery strings': Sending state 16.00000  with 0 decimals of accuracy
[... cut ...]
[19:44:17][W][jnge_modbus:089]: JngeModbus CRC Check failed! C071!=00
[19:44:17][W][jnge_modbus:089]: JngeModbus CRC Check failed! 578F!=96BC
[19:44:17][D][uart_debug:114]: <<< 00:00:00:00:00:01:E9:A1:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FC:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FA:06:04:3A:03:EC:00:A9:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:08:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:BC:96:FF:FF:FF:FF:FD:FD:FF:FF:FF:FF:BF:FF

syssi, 2022-12-07:
Do you use two wires (A+, B-) or three wires (A+, B-, GND) between the RS485 module and your JNGE charger?

syssi, 2022-12-07:
Incoming traffic (extracted from the capture):

```
FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:6F:FF:7F:FD:FF:FF:FF:FF:FF:FF:FB:FF:BF:FF:BF:7F:FF:FF
FF:FF:FF:FE:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:7F:FF:FB:FF:FE:FE:FF:FF:FF:06:03:32:02:70:02:50:02:50:02:40:02:40:02:10:02:20:01:C0:02:00:01:C0:00:02:00:01:00:03:00:06:00:32:00:3C:00:18:00:00:00:01:30:F7:DF:FF:FF:D5:95:17:D0:00:01:95:27:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:F7:FF:FE:FF:FF:FF
FF:FF:FF:FF:FF:FF:FB:FF:FF:DF:BF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:BF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:A6:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:BC:A8:EF:FF:FF:FE:FF:FF:FF:FF:FF
FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:AA:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:06:00:10:00:03:00:01:00:00:00:00:00:01:42:1A:FF:F7:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FB:FE
FF:FF:FE:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:A1:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:E8:4B
EF:FF:FF:FF:FF
FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:06:03:32:02:70:02:50:02:50:02:40:02:40:02:10:02:20:01:C0:02:00:01:C0:00:02:00:01:00:03:00:06:00:32:00:3C:00:18:00:00:00:01:00:00:00:01:00:01:00:10:00:03:00:01:95:27:FF:FF:FF:BF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:FF:FF:06:04:3A:03:EC:00:A2:02:0A:00:00:00:00:00:00:00:09:00
00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:00:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:16:C7:FF:DF:FF:F7:FF:FE:FF:7F:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FF:FD:FF:FF:FF:FF:BF:7F:EE:FF:FF:DF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:7F:FE:FF:FE:FF:FF:FF:FF:FB:FF:FF:FB:FE:FF:FF:EE:FF:FF:FF:FF:06:04:3A:03:EC:00:A3
02:0A:00:00:00:00:00:00:00:19:00:00:00:00:00:28:00:00:00:02:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:00:03:00:05:00:10:00:03:00:01:00:00:00:00:00:01:BD:7C:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:FF:06:04:3A:03:EC:00:A6:02:0A:00:00:00:00:00:00:00:01:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00
00:00:FF:D0:60:00:00:06:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:17:12:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:CE:FD:FF:FA:FE:FE:FF:FF:FE:FB:FF:06:04:3A:03:EC:00:A7:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00
00:01:17:12:FF:FF:FF:FF:FF:FF:FF:FF:FF:CF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FC:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:06:04:3A:03:EC:00:A4:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:E9:96:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EC
FF:7F:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:FF:06:04:3A:03:EC:00:A1:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:E8:4B:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:E9:FD:BE:FF:FF:FF:FF:FB:FF:FF:FF:FF:06:04:2A:03:EC:00:AA
02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:EF:B0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:42:1A:FF:FF:FB:FF:FF:FD:FF:FF:FF:FF:FB:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F8:F7:FE:FF:FF:7F:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:AA:02:0A:00:00:00:00:00:00:00:08:00:00:00:00
00:08:00:00:00:03:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:03:40:1A:FF:FF:FF:FF:BF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FE:FF:FE:FF:FF:FF:FF:FF:FF:06:04:3A:03:EC:00:AA:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00
10:00:03:00:01:00:00:00:00:00:01:42:1A:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FD:F4:7F:FE:FF:FF:FF:FF:FF:FF:FF:06:04:38:01:EC:00:AA:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:42:1A:DD:FF:FD
FF:FD:FF:FF:FF:FF:FF:FF:FD:FF:FD:FD:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FF:BF:FF:FF:06:04:3A:01:EC:00:A6:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:BC:A9:7F:FF:FF:FB:FF:FF:FF:FF:FF:FF:FA:FE:FF:FD:FE
FF:FF:FF:FF:FF:FE:FF:7F:FF:3E:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:BF:FF:06:04:3A:03:EC:00:A6:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:B0:60:00:00:02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:BC:A9:FF:FF:EF:3F: [... cut ...] F:FA:06:04:3A:03:EC:00:A3:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00
02:02:03:00:07:00:10:00:03:00:01:00:00:00:00:00:01:BD:7C:FF:FF:FF:FF:FF:FF:FF:FF:FF:5F:FF:FF:FF:FF:F9:FF:7B:EF:FF:FF:FF:FF:7F:7F:FF:FF:BF:FD:FF:FF:FF:FF:7F:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:06:04:3A:03:EC:00:AB:02:0A:00:00:00:00:00:00:00:09:00:00:00:00:00:28:00:00:00:01:00:00:00:00:00:00:00:05:00:00:00:00:FF:D0:60:00:00:02:02:03:00:07:00:10:00:03:00:01
```

syssi, 2022-12-07:
The noise between the frames isn't normal IMO. Please try to connect `GND` between the MPPT charger and the RS485 converter.

woodie2, 2022-12-08:
> Do you use two wires (A+, B-) or three wires (A+, B-, GND) between the RS485 module and your JNGE charger?

yes I have connected all 3 wires.

syssi, 2022-12-13:
If you have the chance please replace the RS485 converter and the cable between the inverter and your ESP. The posted protocol datasheet is the one I did use to implement this component. The `FF:FF:FF:FF:FF:FF:FF:FF:FF:FF` bytes look like faulty hardware or bad grounding.

woodie2, 2023-01-01:
the RS485 converter was the problem, I have now a other charge of them, and it works without error.

syssi, 2023-02-05:
Could you do me a favor? Your JNGE MPPT controller uses the new RJ45 pinout of the RS485 port, right? Could you tell me the manufacturing date (sticker on the device) and the model name? I would like to improve the documentation and I'm looking for some easy why to identify the pinout.

commit f8344b73 (2023-02-06): Add new RJ45 pinout (Closes: #35) (#37)
closed 2023-02-06 by syssi